jallib/jallib #202: Adaptation of device files of 16f183x3 for ADC libraries

robhamerling, 2015-07-26:
With the 16f183x3 ADC pins are differently named/numbered than before: pin_ANx aliases are not anymore sequentially named representing the ADC channel number. For example the 16f18323 has pin_ANA0..2, and pin_ANA4..5,  and pin_ANC0..5.  The bits of the ANSEL registers corresponding to the ADC-pins are similarly named.  The corresponding ADC channel numbers are 0-2, 4-5 and 16-21 (the value to be specified in ADCON0_CHS). 
Although this naming convention seems more logical than so far, the ADC libraries may not work with these names. There is probably no problem with the pin names and channel numbers, but possibly with the ANSEL bits.  In the past we adopted  a "JANSEL_ANSx" alias naming convention of the ANSEL bits, whereby 'x'  corresponds to the channel number. To avoid a (major) redesign of the ADC libraries we would get in this case JANSEL_ANS0,1,2,4,5 and JANSEL_ANS16,17,18,19,20,21. 
For example for channel 18 of the 16f18323 this would become:
   var volatile bit    ANSELC_ANSC2              at ANSELC : 2
   alias                  JANSEL_ANS18                is ANSELC_ANSC2
Seems the way to go, but it must be checked if the ADC libraries will still work with these names.

robhamerling, 2015-07-29:
I think one of the issues is that the ADC libraries assume that the number of ADC channels (the const byte ADC_NTOTAL_CHANNEL = xx in the device files) corresponds to the highest channel number.  That will not (always) be the case with the 16f183xx. For example the 16f18345 has 17 ADC channels, but the highest channel number is 23.

RobJansen62, 2019-04-21:
See issue #229.

closed 2019-04-21 by RobJansen62