jallib/jallib #203: GLCD_COMMON glcd_line procedure: not working for vertical lines

derousrian, 2015-08-30:
I use the GLCD_KS0108 & GLCD_COMMON library's to control the GLCD on my Easypic V7 devboard..... When i use the GLCD_LINE procedure for a vertical line, then something goes wrong.... 

I see only a few dots....

mattschinkel, 2015-08-30:
Hi,

Please discuss this at https://groups.google.com/forum/#!forum/jallib

Please post your code for us so we can assist. You may need to do your own troubleshooting of the glcd_line() procedure.

Make sure glcd_write_pixel() is working and glcd_pen_color is set before using this line procedure.

Matt.

closed 2015-08-30 by mattschinkel

robhamerling, 2015-09-03:
On 2015-08-30 13:31, derousrian wrote:
>
> I use the GLCD_KS0108 & GLCD_COMMON library's to control the GLCD on
> my Easypic V7 devboard..... When i use the GLCD_LINE procedure for a
> vertical line, then something goes wrong....
>
> I see only a few dots....
>
>

I've here a test board with a 18f2580 and a KS0108 glcd. A simple test 
program which writes a number of vertical lines shows nice complete 
vertical lines.

> glcd_clear_screen()
>
>    for 32 using i loop
>       glcd_line(4 * i, 0, 4 * i, 63)
>       delay_100ms(2)
>    end loop

See attached picture.

In the past I had some trouble with this setup (distorted graphics). It 
appeared to be a broken PCB trace of one of the data pins.
It sounds to me you have a similar problem. Long wires between PIC and 
display can also be problematic.

Good luck!
Regards, Rob.

-- 
*Rob H*amerling - http://www.robh.nl