jallib/jallib #205: I2C for PIC16F18855

igninja, 2016-02-04:
I2C libraries don't work with new Micro-controllers of 188xx family

robhamerling, 2016-08-10:
I'm not an experienced I2C user, but I have a project with a 16F18855 as I2C slave.
The 18f18855 uses a different PIE/PIR pair for the SSP interrupt than the library assumes. To get the I2C library working I only had to add to my program the following 2 lines:
```
   alias PIR1_SSPIF is PIR3_SSP1IF              -- device difference
   alias PIE1_SSPIE is PIE3_SSP1IE
```
This is a circumvention for an omission in the library. A more elegant solution is to let the library check which PIE/PIR pair to use for the actual PIC.

RobJansen62, 2020-06-22:
This was fixed some time ago.

closed 2020-06-22 by RobJansen62