jallib/jallib #208: PWM library update

robhamerling, 2016-05-29:
The PWM libraries (in particular PWM_PWMx) need be updated to support the PWM facilities of newer PICs like 16f15xx, 16f17xx and 16f18xx(x) series:
(1) Support for PWM channels higher than channel 4: some PICs have channel 12 
(2) Support for PWM channels with 16-bits resolution

robhamerling, 2016-12-17:
Another enhancement is desired for the 12f1840 and maybe other similar PICs. 
The current libraries use a fixed  setting of ccp1con_ccp1m (0b1111). But the 12f1840 supports a choice of output polarities by means of other patterns for ccp1con_ccp1m. Unfortunately with the current PWM libraries the PWM output polarity of the 12f1840 is the reverse of the polarity with the 'classic' PICs (the datasheet calls it 'active low').

Multitrain, 2019-06-30:
In JAL V2 PWM library  an error in the pwm_ccp1.jal  makes the PWM output of PIC inverted. Even it worksOK for "old PICs" not supporting output polarity inversion like 16F87x, for the newer with Enhanced PWM like the 16F88x you have to take care of the issue. For simple PWM with an Enhanced PWM, follow that change

To override this, change line 43  
_ccp1con_shadow_ccp1m = 0b1111
to
_ccp1con_shadow_ccp1m = 0b1100

RobJansen62, 2019-07-01:
> In JAL V2 PWM library  an error in the pwm_ccp1.jal  makes the PWM output of PIC inverted. Even it worksOK for "old PICs" not supporting output polarity inversion like 16F87x, for the newer with [...]

I only see this error in pwm_ccp1.jal and not in the others (pwm_ccp2.jal to pwm_ccp10.jal), right?

robhamerling, 2019-07-01:
The PWM libaries (pwm_ccpx and pwm_pwmx) are generated with a script from templates, see the pwm subdirectory in tools.  So changes should be made in the templates and then new libraries should be generated from the modified templates, which will ensure consistency! An individual change of a generated library is possible, but will be undone when new libraries are generated from the templates!

RobJansen62, 2019-07-01:
Hi Rob,
I did not know that. I fixed the issue manually and tested the fix on a PIC16F73 and a PIC16F882 and it seems to work. There is however a difference in the behavior when running the sample program (I created a new sample program 16f882_pwm_led.jal) since when calling pwm1_off the LED in the new sample progra is on because of the inverted output. I will have a look later at the templates you mentioned.

RobJansen62, 2019-07-01:
Hi Rob,

I checked the template file and there the value is already 0b1100. I saw earlier that only pwm_ccp1.jal had the wrong value. All others have the correct value. Not sure why that  is.

Kind regards,

Rob

RobJansen62, 2024-12-28:
Fixed. Will be part of the jallib release following jallib version 1.9.0.

closed 2024-12-28 by RobJansen62