jallib/jallib #209: PIC18F46K22 device file - wrong pin definition, component of SPI2 peripheral

funlw65, 2016-06-02:
**Instead of** 
`alias               pin_SD02_direction        is pin_D4_direction`
**it should be**
`alias               pin_SDO2_direction        is pin_D4_direction`

_The error may be on the original Microchip header..._

sstandfast, 2016-06-02:
Maybe I'm just still asleep but I don't see the difference between the two?
Shawn

funlw65, 2016-06-02:
instead of 0(zero) must be O (letter O capital.),
From SPI DataOutput, not Data0 (zero).

robhamerling, 2016-06-02:
On 06/02/2016 02:23 PM, Vasi wrote:
>
> *Instead of*
> |alias pin_SD02_direction is pin_D4_direction|
> *it should be*
> |alias pin_SDO2_direction is pin_D4_direction|
>
> /The error may be on the original Microchip header.../
>
You are right! It is an error in the MPLAB-X file from which the device 
files are generated.
BTW The pin definition itself is also in error (pin_SD02 -> pin_SDO2).
These errors are in a number of device files (16f1829, 18f43/4/6k22 and 
their LF variants),
Microchip should be informed.

Regards, Rob.

>
-- 
*Rob H*amerling - http://www.robh.nl

robhamerling, 2016-06-04:
On 06/02/2016 02:23 PM, Vasi wrote:
>
> *Instead of*
> |alias pin_SD02_direction is pin_D4_direction|
> *it should be*
> |alias pin_SDO2_direction is pin_D4_direction|
>
> /The error may be on the original Microchip header.../
>
It could be easily fixed by adding the lines:

>  elif (alias == "SD02"):
>        aliaslist.append("SDO2")
>         print "  Replaced", alias, "by SDO2 for pin", pinnumber
in the file pinmap_create.py in the tools directory (e.g. after line 91)
and then generate new device files as with a new MPLABX version.

Regards, Rob.

-- 
*Rob H*amerling - http://www.robh.nl

robhamerling, 2017-11-12:
Solved

closed 2017-11-12 by robhamerling