jallib/jallib #218: Include file for 16F1847 Pin_A5 not defined for input

FraserSmith51, 2017-09-02:
Title above says it all. As it stands at the moment compiling with the existing include file for the 16F1847 gives a pin not defined error when trying to use pin_a5 for input and with "pragma target MCLR   internal             -- reset externally"
I believe it needs 
var volatile bit    pin_A5                    at PORTA : 5
alias               pin_SS1_RA5               is pin_A5

but where do these lines go?

robhamerling, 2017-09-03:
Technically the place is not very important, but for consistency with similar situations (MCLR at pin_A3 or pin_E3 with other PICs) I suggest to put these lines immediately after the declaration of PORTA_RA5.

robhamerling, 2017-11-12:
Solved