jallib/jallib #221: datasheet wikis fail

robhamerling, 2018-03-08:
The links in the datasheet wikis don't work anymore for newer PICs. 

Until recently the filename of the PDF file at the site of Microchip consisted of a 5 or 8 digits number + a suffix (letter). But of newer PICs the PDF files have a 'descriptive' filename. It looks like but is not exactly the same as its title, so the 'simple' URLs in the wikis result in a 404 error message.

Maybe the URL could be replaced by a search argument.

RobJansen62, 2018-03-11:
Could we not upload the datasheets on jallib?

robhamerling, 2018-03-11:
Even when possible and allowed I do not like this kind of redundancy and related extra work. 
The URLs in the wikis were a convenience, but without URLs people can retrieve datasheets themselves pretty easily on the site of Microchip.

RobJansen62, 2018-10-22:
Wikis have been deleted by Rob Hamerling. Pythin script for creating wikis were moved to archive/tools.

closed 2018-10-22 by RobJansen62