jallib/jallib #225: wrong data memory specifications for 16f57 and 16f59

robhamerling, 2018-11-24:
With repect to the device files of level MPLABX 5.0 

- For the 16f57 'pragma shared' specifies 0x8-0x0F, which should be 0x0-0xF

- For the 16f59  'pragma shared' specifies 0xA-0x0F, which should be 0x0-0xF
  and 'pragma data' specifies too many data memory locations (compiler supports max 4 banks!)

RobJansen62, 2018-11-25:
Thanks for the info. I have some questions related to this.
Should shared for 16f57 not be 0x0 to 0x7 since data is already from 0x8 to 0x1F?
Should shared for 16f59 not be 0x0 to 0xA since data is already from 0xA to 0x1F.
Is the compiler limitation of 4 banks a compiler problem or an error in the datasheet?

robhamerling, 2018-11-28:
There is much more wrong with data memory declarations with the 16f59
SFRs occupy range 0x0-0x9  - OK
_pic_accum and _pic_isr_w occupy range 0xE-0xF  - OK
PORTA-shadow to PORTD shadow occupy range 0xD-0xA - OK 
PORTE_shadow is declared at 0X9 - occupied!
TRISA_shadow - TRISC_shadow are declared in range 0x8-0x6 - occupied!
OPTION_REG_shadow at 0x5 - occupied!
TRISD-shadow - TRISE_shadow at range 0x4-0x3 - occupied!

And in the Miscellaneous information at the end of the device file:
--  Free shared memory: 0x0-0x2 - wrong!

Conclusion: the allocation of shadow bytes uses a wrong range of shared memory.
I'll try to find out where it goes wrong.

robhamerling, 2018-11-29:
Sorry, I must correct myself.
 I installed MPLABX IPE version 5.10 for private use and out of habit I generated device files from this version. The observations about occupied memory for shadow registers are based on these device files, not on those of the current Jallib device files. 
However I used the pic2jal.py script in Jallib/tools and it seems that the shared memory specification is OK with MPLABX 5.10, but the memory allocation of shadow registers seems to need revision (using only 'free' shared memory).

robhamerling, 2018-12-01:
The 'archive/README.txt' file at the site of Kyle York may say that the 16f54/57/59 are 'known not to work'. But we have proven that these PICs do work, with the exception that of the 16f59 only data memory banks 0-3 are supported. When the device file specifies only 4 memory banks I see no reason why to remove the 16f59 from Jallib. See also issue #19.

RobJansen62, 2018-12-04:
The data space for the 16f57 seems OK. The data space for the 16F59 has been limited to 4 banks by adding a new data space to devicespecific.json.
So the original space: 0xA-0x1F,0x30-0x3F,0x50-0x5F,0x70-0x7F,0x90-0x9F, 0xB0-0xBF,0xD0-0xDF,0xF0-0xFF
Has been limited in devicspecific.json to:  0xA-0x1F,0x30-0x3F,0x50-0x5F,0x70-0x7F
This limits the amount of usable data space and is a workaround for the compiler limitation.

closed 2018-12-04 by RobJansen62