jallib/jallib #226: typo in usb_drv.jal at usb_sdp_request_type

PeterHolik, 2019-03-26:
In commit "replace var is by alias for _af board files and their derived samples"

https://github.com/jallib/jallib/commit/9bf1d651cc645219c039cee05ed84f2d90741ccf#diff-6cc8329f22cf5c829b8bc5bb2a4a1a78

i found a typo
--- usb_drv.jal.orig	2019-03-26 14:51:24.194911438 +0100
+++ usb_drv.jal	2019-03-26 08:52:14.294055766 +0100
@@ -462,7 +462,7 @@
 				
 				
 				-- obtain request
-				case ( usb_sdp_request_type  & 0b_0110_000 ) of
+				case ( usb_sdp_request_type  & 0b_0110_0000 ) of
 					(0b00 << USB_REQUEST_TYPE0_BIT ):
 					block
 						-- serial_hw_write( "S" )

RobJansen62, 2019-03-26:
Hi Peter,
Good that you found it. I am not sure what the compiler uses when it parses 0b_0110_000. I can modify it and see if the driver still works OK. Did you encounter any problems?
Kind regards,
Rob

PeterHolik, 2019-03-27:
I checked what the compiler is doing with 0b_0110_000 or 0b_0110_0000.
The asm differs 0b_0110_000 was 48 and 0b_0110_0000 was 96.

I had to patch because i use " usb_handle_vendor_request_callback" and it didn't work without.

RobJansen62, 2019-03-27:
Hi Peter,

Good to know. Thanks for the info. I will make an update soon.

Kind regards,

Rob

RobJansen62, 2019-03-30:
Fixed.

closed 2019-03-30 by RobJansen62