jallib/jallib #234: suspicious output at usb_serial

PeterHolik, 2019-12-16:
To debug reading of a DS18S20 1-wire sensor i use an 18f14k50.

The output made me crazy with suspicious charactes and restarts.

I tried to comile against an older jaillib and did not see this.

At the end i found out the origin was in jallib/include/device/18f14k50.jal

after this small change every thing worked like it should

-pragma  data               0x00-0x2FF
+pragma  data               0x60-0x2FF

RobJansen62, 2019-12-16:
Hi Peter,
If I look at the datasheet then the address 0x00-0x2FF should be OK. There is a special area called 'access bank' in address range 0x0-0x5F,0xF60-0xFFF but this should work OK. So if I would use a DS18D20 with a PIC18F14k50 (and the original range of pragma data 0x00-0x2FF) then I should have this problem right? Of which date was your older device file?

PeterHolik, 2019-12-16:
I tested with 18f14k50.jal from 20. Nov 2011.

Again my problem is the strange output of usb_serial.

RobJansen62, 2019-12-17:
Hi Peter, 
Can you send your sample program to the jallib google group so that I can use it to reproduce it otherwise I have to set up the whole program and that takes me much more time. I had a look at an old jallib release of 2011 that you mentioned and saw the following differences in the device file as you already had noticed. 
2011:
pragma  data    0x60-0xFF,0x100-0x1FF,0x200-0x2FF
pragma  shared  0x0-0x5F
2019:
pragma  data               0x0-0x2FF
pragma  shared             0x0-0x5F,0xF60-0xFFF
According to the datasheet the 2011 is - as I can see now - not correct since it gives you less data space (2011: 672 bytes, 2019: 768 bytes) but still it is strange that it does not work with the new device file. It could also be a compiler issue (hopefully not).

PeterHolik, 2019-12-17:
It is sufficient to print something out.

`-- compile: ../jallib/compiler/jalv2-x86-64 -loader18 -no-fuse  -s ../jallib/include/jal -s ../jallib/include/peripheral/memory_pointer \
--                                          -s ../jallib/include/device/ -s ../jallib/include/peripheral/usb 18f14k50_usb_serial.jal
-- program: sudo ../fsusb/fsusb-0.1.11-2/fsusb --program 18f14k50_usb_serial.hex
--          sudo ../fsusb/fsusb-0.1.11-2/fsusb --reset

;@jallib section chipdef
-- chip setup
include 18f14k50
pragma target clock       48_000_000
WDTCON_SWDTEN = OFF                          -- WDT off

const byte str_welcome[] = "JALLIB USB Serial\r\n"

include delay

enable_digital_io()

include usb_serial
include print

usb_serial_init()

var word temp

T1CON_TMR1ON = on

forever loop
   usb_serial_flush()

   if ( usb_cdc_line_status() !=  0x00 )  then
     print_string( usb_serial_data, str_welcome )
     usb_serial_data = " "
     temp = TMR1
     print_word_dec(usb_serial_data, temp)
     print_crlf(usb_serial_data)
     delay_1s(1)
   end if
end loop
`

`Willkommen zu minicom 2.7.1

Optionen: I18n 
Übersetzt am May  3 2018, 15:20:11.
Port /dev/ttyACM0, 13:59:45

Drücken Sie CTRL-A  Z für Hilfe zu speziellen Tasten                                                                           
09                                                                                                                             
                                                                                                                               
                                                                                                                               
2�                                                                                                                        ----+
JALL USB LIB USSeria 5B 
23SB Serl
 1m2
     rial
2909
                XB Serial
 1290
      JAL

         909
JAIB USALLIB B Ser Minicom USB Sial

11
        XLLIB USB Serial 1290m2      l
 129
J

 m2�
        JALLIUSB SIB USBerial 00 4B MB Seri
 12m2
ial
 909

9

al
 274

��
JALL USB LIB USSeril 00 4B MB Seri
 19�sial
`

RobJansen62, 2019-12-17:
Hi Peter,

I saw you hardly defined any pragmas assuming the defaults of the device file but that may not always work. If you look at the sample program for usb serial it defines the pragmas below (I commented two out since they were not present in the old device file but you an enable them for the current device file). Especially the pragma XINST should be set to disabled since the JAL compiler does not support the extended instruction set. The default of the device file is that it is enabled and I do not know what the effect is if you enable it.

I know the usb serial sample file works correctly since I used it several times when developing the usb driver variant for the 16F1455. Can you give it try defining the pragmas below and see what happens? These pragmas assume a crystal of 12 MHz.

Thanks

pragma target clock       48_000_000

-- fuses
pragma target CPUDIV        P1              -- NO_CPU_SYSTEM_CLOCK_DIVIDE
pragma target USBDIV        P1              -- USB_CLOCK_COMES_DIRECTLY_FROM_THE_OSC1_OSC2_OSCILLATOR_BLOCK_NO_DIVIDE
pragma target OSC           HS
--pragma target PLLEN         ENABLED         -- OSCILLATOR_MULTIPLIED_BY_4
pragma target FCMEN         DISABLED
pragma target IESO          DISABLED
pragma target PWRTE         DISABLED        -- power up timer
pragma target BROWNOUT      DISABLED        -- no brownout detection
pragma target VOLTAGE       V30             -- brown out voltage
--pragma target WDT           CONTROL         -- watchdog software controlled
pragma target WDTPS         P32K            -- watch dog scaler setting
pragma target MCLR          EXTERNAL        -- external reset
pragma target LVP           DISABLED        -- no low-voltage programming
pragma target XINST         DISABLED         -- extended instruction set
pragma target DEBUG         DISABLED        -- background debugging
pragma target CP0           DISABLED        -- code block 0 not protected
pragma target CP1           DISABLED        -- code block 1 not protected
pragma target CPB           DISABLED        -- bootblock code not write protected
pragma target WRT0          DISABLED        -- table writeblock 0 not protected
pragma target WRT1          DISABLED        -- table write block 1 not protected
pragma target WRTB          DISABLED        -- bootblock not write protected
pragma target WRTC          DISABLED        -- config not write protected
pragma target EBTR0         DISABLED        -- table read block 0 not protected
pragma target EBTR1         DISABLED        -- table read block 1 not protected
pragma target EBTRB         DISABLED        -- boot block not protected
pragma target HFOFST        ENABLED         -- THE_SYSTEM_CLOCK_IS_HELD_OFF_UNTIL_THE_HFINTOSC_IS_STABLE

WDTCON_SWDTEN = OFF                          -- WDT off

PeterHolik, 2019-12-18:
There is no differences in the compiled output (.asm)  and behavior.

RobJansen62, 2019-12-18:
There is a difference in the fuses (check the hex files) but I do not know if that will impact the behavior but apparently not since you tested it. I will try something this weekend.

RobJansen62, 2019-12-19:
I noticed something in the above program. There is a delay of 1 second and that may be too long since the usb_serial_flush() is not called frequently enough for the USB to function normally. Please try using a smaller delay (e.g. 500 us) and a counter for extending the time.

PeterHolik, 2019-12-20:
ok changed delay_1s(1) to

`     for 2000 loop
       delay_10us(50)
       usb_serial_flush()
     end loop`

but same behavior.

The diff of an asm file from none working to working version gives a lot of
changes from v__access to v__banked like for example

-                               movwf    v__pic_temp,v__access
+                               movwf    v__pic_temp,v__banked

RobJansen62, 2019-12-20:
I performed the following tests:
1) Running your original program with only 1 pragma --> My computer did not recognize the USB serial port.
2) Running your original program with all pragmas set --> My computer did recognize the USB serial port but no data was transmitted
3) Same as 2) but then removing the 1 sec delay and making a loop that services the USB driver every 1 ms --> Program worked.

So I think there is something wrong with your set-up. Final program given below.

-- chip setup
include 18f14k50
pragma target clock 48_000_000

-- fuses
pragma target CPUDIV        P1              -- NO_CPU_SYSTEM_CLOCK_DIVIDE
pragma target USBDIV        P1              -- USB_CLOCK_COMES_DIRECTLY_FROM_THE_OSC1_OSC2_OSCILLATOR_BLOCK_NO_DIVIDE
pragma target OSC           HS
pragma target PLLEN         ENABLED         -- OSCILLATOR_MULTIPLIED_BY_4
pragma target FCMEN         DISABLED
pragma target IESO          DISABLED
pragma target PWRTE         DISABLED        -- power up timer
pragma target BROWNOUT      DISABLED        -- no brownout detection
pragma target VOLTAGE       V30             -- brown out voltage
pragma target WDT           CONTROL         -- watchdog software controlled
pragma target WDTPS         P32K            -- watch dog scaler setting
pragma target MCLR          EXTERNAL        -- external reset
pragma target LVP           DISABLED        -- no low-voltage programming
pragma target XINST         DISABLED         -- extended instruction set
pragma target DEBUG         DISABLED        -- background debugging
pragma target CP0           DISABLED        -- code block 0 not protected
pragma target CP1           DISABLED        -- code block 1 not protected
pragma target CPB           DISABLED        -- bootblock code not write protected
pragma target WRT0          DISABLED        -- table writeblock 0 not protected
pragma target WRT1          DISABLED        -- table write block 1 not protected
pragma target WRTB          DISABLED        -- bootblock not write protected
pragma target WRTC          DISABLED        -- config not write protected
pragma target EBTR0         DISABLED        -- table read block 0 not protected
pragma target EBTR1         DISABLED        -- table read block 1 not protected
pragma target EBTRB         DISABLED        -- boot block not protected
pragma target HFOFST        ENABLED         -- THE_SYSTEM_CLOCK_IS_HELD_OFF_UNTIL_THE_HFINTOSC_IS_STABLE

WDTCON_SWDTEN = OFF                          -- WDT off

const byte str_welcome[] = "JALLIB USB Serial\r\n"
include delay
enable_digital_io()
include usb_serial
include print

usb_serial_init()

var word temp
T1CON_TMR1ON = on

forever loop
   for 1000 loop
      usb_serial_flush()
      _usec_delay(1_000)
   end loop
   if ( usb_cdc_line_status() != 0x00 ) then
      print_string( usb_serial_data, str_welcome )
      usb_serial_data = " "
      temp = TMR1
      print_word_dec(usb_serial_data, temp)
      print_crlf(usb_serial_data)
   end if
end loop

RobJansen62, 2020-01-02:
Closed since not reproducible.

closed 2020-01-02 by RobJansen62

PeterHolik, 2020-01-07:
Sorry for the noise, i found now out what happens:

I used 18f14k50_usb_bootloader_autostart.jal from 22. Dec 2011 with 

XINST         ENABLED

then usb serial works correct with pragma data 0x60

recent 18f14k50_usb_bootloader_autostart.jal has

XINST         DISABLED

then usb serial works also with pragma data 0x0

Sorry again.

robhamerling, 2020-01-07:
Hi Peter,

I found a section in the datasheet of the 18F14K50 which applies probably also to other 18Fs, see below. This seems to explain the cause of your troubles. Kyle York (original author of the JalV2 compiler) was very strict: "XINST must be disabled"

Regards, Rob.

From datasheet 40001350F section 25.2.4:
_Additionally, the Indexed Literal Offset Addressing
mode may create issues with legacy applications
written to the PIC18 assembler. This is because
instructions in the legacy code may attempt to address
registers in the Access Bank below 5Fh. Since these
addresses are interpreted as literal offsets to FSR2
when the instruction set extension is enabled, the
application may read or write to the wrong data
addresses._