jallib/jallib #236: Task Feature

kozmotronik, 2021-01-25:
Hi Team!
JAL has the Tasking feature by version 2. But the information about tasks in the compiler guide is quite minimal. Could any one add an example using task feature in the sample folder please?

RobJansen62, 2021-01-25:
I think it would have been better to post this on the Jallist of Jallib Google Group but I found one example. I will create a sample program later that will make it visible that it works (this example does not show it).

VAR VOLATILE BYTE counter1
VAR VOLATILE BYTE counter2

TASK task1(BYTE in aa) is
   counter1 = aa
   FOREVER LOOP
   counter1 = counter1 + 1
   SUSPEND
   END LOOP
END TASK

TASK task2(BYTE in aa) is
   counter2 = aa
   FOREVER LOOP
   counter2 = counter2 + 1
   SUSPEND
   END LOOP
END TASK

START task1(10)
START task2(20)

FOREVER LOOP
   SUSPEND
END LOOP

kozmotronik, 2021-01-25:
Thanks for your kindly response and sorry for bothering if here is not where I had to ask this. I am so excited with  JAL especially when I have seen the tasking feature. It seems a very handy cooperative multitasking management for PIC micros.  JAL really has become a solid rock compiler this all thanks to you developers. I will seek to post this in Jallist then.

kozmotronik, 2021-01-25:
Hi Rob, unfortunately I could not manage how to reach the Jallist. I did managed to find the jallip google group but not the jallist. There is only a link to the old yahoo jallist group but it is not available anymore. I also would like to contribute for JAL, you know, it may be any thing to handle. I am good at C programming. I am very familiar with PIC architecture and PIC assembly. I don't know what can I contribute for, but I do want to contribute. In the past I tought about writing a modbus RTU library for JAL, but I could not complete its tests because of a lot of work load. But I still have that uncompleted code.

RobJansen62, 2021-01-26:
Hi Kozmotrink,
You have to go to Google Groups. The Jallib Google group is used for issues with libraries and the compiler. The Jallist Google group is used for users that have questions about using JAL. Both Links are provided on the readme.md page. It would be nice to have more contributors. The Yahoo group is no longer used since Yahoo stopped with the Groups. 
FYI. I made a small different test program with tasks and got some compiler issues. I am not yet sure what is going on. Will do some more investigation this weekend. I will close this issue once I have a working sample program and made it part of Jallib.

kozmotronik, 2021-01-26:
I see, I will have a look at readme.md page for Jallist thanks. JAL is dramatically powerful with all its libraries and integrations thanks to all its developers and I want it to be more popular among people. That's why I have been writing articles in Turkish in [this site](http://bilgiveuygulama.com/index.php?git=23). As I said earlier I'd like to contribute as well. But I don't know where to start to contribute and what to do. I am Turkish, thus I am writing articles for the Turks, using jalv2LanguageReference.pdf as a source and this is one of the ways of contribution. I can go further by continuing the development of MODBUS RTU library and then release for you to revise if it is ok to add into standart Jallib libraries. By the way I will look forward for the new tasking examples.
Best Regards.

RobJansen62, 2021-01-26:
Hi Kozmotronik,
That is nice. I also try to promote JAL by putting my project on Instructables. Good to see that you are doing the same thing. If you want to contribute there is a Wiki that explains something about the coding rules. Once you do that you always have to provide at least one sample program with your library. The JAL coding rules are checked by a python script. I can help you with it once you created this (maybe I should add that to the wiki page too). I made an actual test program (attached as zip) and it works but one thing puzzles me and that is step 4) in the description of the sample program. This sample program at least shows that the task switching actually works. I will add this to Jallib once I can explain step 4). 
[12f617_tasks.zip](https://github.com/jallib/jallib/files/5875342/12f617_tasks.zip)

kozmotronik, 2021-01-27:
Hi Rob,
I have tested your task sample with PIC16F1509. In this test I observed that the task blue works double after the first execution. I mean when it is executed first, the LED blinks 3 times and in all the other executions LED blinks 6 times. Somehow the task blue is being executed twice successively. Although I don't know how the compiler manages the tasks exactly, I was asking to myself if it is a good idea giving task management to the compiler and how flexible it can be? In your example you used software delays and it blocks the CPU from executing the other tasks. So how non-blocking calls can be implemented using task feature. I had researched about [Duff's Device](https://en.wikipedia.org/wiki/Duff%27s_device) and some other multitasking techniques including FreeRTOS. I have rewritten and implemented [Protothreads](http://dunkels.com/adam/pt/index.html) and even tried to rewrite the FreeRTOS completely but FreeRTOS is very heavy for the 8-bit PICs. Then I understand that the best way to multitask on 8-bit PICs is doing it without a pre-emptive schedular which needs a lot of memory and context switching. Protothreads that based on Duff's Device technique is best suited for this type of multitasking. Unfortunately this technique uses C's macro feature. I will work around to figure out how it can be adopted to JAL.

RobJansen62, 2021-01-28:
Hi Kozmotonik,
I updated the task sample program using a timer 0 library so that there are no blocking calls.. This works fine. There is one strange thing and that is the control of the led. In the sample program I use a variable to keep track of the led value. If I, however, use the pin to toggle the led only the red led works but the others stay on. 

So this works:
led_value = !led_value 
led = led_value

But this not:
led = !led
The later is strange since I why does it work for the red led bu
[12f617_tasks.zip](https://github.com/jallib/jallib/files/5889724/12f617_tasks.zip)
t not for the others?

kozmotronik, 2021-01-31:
Hi Rob,
Good to hear from you. I will test your new updated example. By the way, I had modified your previous example to make it non blocking, but I think I failed. Surely one of the reasons of my failure is because I still don't have enough knowledge about the tasks and how they behave exactly. You can get that code from [here](https://drive.google.com/file/d/1rLBequvGw3VGSvVvYp_qu9Vf0hgRilRP/view?usp=sharing) if you'd like to check it out. Actually that code works, but not as expected. What happens is tasks don't yield until they finish the for loop. But the idea was that they yield if the delay time is not up. The issue that you mentioned: "If I, however, use the pin to toggle the led only the red led works but the others stay on." might be related to this. Then I had a look at the assembly output file to figure out how it works exactly but it is a little hard figuring out just checking the assembly code. For these reasons I have 3 questions:

1. What happens exactly when the suspend is executed?
2. Is suspend has the manner of yield which a task gives the control of the CPU to another task?
3. How could we implement a task that making IO operations and it yields the control if the IO is not ready yet? 

Best regards,
İsmail

RobJansen62, 2021-02-01:
Hi Kozmotronik,
I checked your program but a few things should be changed:
1) You use the same delay slots for all leds and I think you should use different delay slots. You define DELAY_SLOTS = 3 so ds_red, ds_green and ds_blue should not both be 0 but 0,1 and 2.
2) Your suspend outside of the forever loop will never be called. Please stick to the original program. I added a version which also uses the LED pins.
Suspend releases control from the task and then the next task will start. There is some code created by the compiler that determines which task gets control in the order the tasks where started. After all tasks got their turn, it starts again with the first task (which is the main program which is also a task). 
If tasks use the same IO you have to make your own precautions to see if it is already in use by another task.
Have a look at this video that I made this weekend: https://www.youtube.com/watch?v=tOFy_s7aVnQ&feature=youtu.be
[12f617_tasks.zip](https://github.com/jallib/jallib/files/5905682/12f617_tasks.zip)

kozmotronik, 2021-02-01:
> 1. You use the same delay slots for all leds and I think you should use different delay slots. You define DELAY_SLOTS = 3 so ds_red, ds_green and ds_blue should not both be 0 but 0,1 and 2.

Thank you Rob. Yeah, I am the victim of <copy - paste> because of my eagerness :laughing:. I have already looked at your video and I am the one who commented below with the nick of isotürk. In the video I noted the LEDs blinks well without blocking each other. But that issue you commented:
> So this works:
led_value = !led_value
led = led_value

>But this not:
led = !led

is really interesting. I adopted your last example for 16f1509 and replacae led_values with the pin aliases, I got them worked.  May be it is an issue specific to the 12f617. Could there be a problem with changing banks in the compiler?

calpgithub, 2021-02-01:
Hallo

These delay_slot and delay_check  are also jal  ? You gotl some more info a about it  ? In the manual or books I had  not found them.

Thanks

Bart

calpgithub, 2021-02-01:
Hallo

Got it in the libraries....

Bart

kozmotronik, 2021-02-01:
> Hallo These delay_slot and delay_check are also jal ? You gotl some more info a about it ? In the manual or books I had not found them. Thanks Bart

Hi Bart,
Delay slots and check delay belong to "timer0_isr_interval.jal" library which you can find it where you installed Jallib and in the "lib" folder. If you don't installed Jallib yet you can have look at that library from: https://github.com/jallib/jallib/blob/master/include/peripheral/timer/timer0_isr_interval.jal. This library is for making asyncronous non-blocking periodic delays, alternative to blocking _usec_delay and other delay functions that resides in "delay.jal" library. If you want to make use of it, just have a look at the timer0_isr_interval.jal file and you will find an example about how to use it.

RobJansen62, 2021-04-05:
A video was made on this topic. Issue can be closed.
https://youtu.be/tOFy_s7aVnQ

closed 2021-04-05 by RobJansen62